micropython-IMU/micropython-mpu9x50 #4: missing pyb import when run test

CharlesCCC, 2016-09-17:
missing pyb import when run test code.

peterhinch, 2016-09-19:
The pyb module is specific to the [Pyboard](http://micropython.org/). These drivers were developed for this platform.

If you are running the code on a Pyboard, what is the exact error message you are seeing? Otherwise please let us know your platform - we might consider doing a port.

CharlesCCC, 2016-09-19:
@peterhinch Got you. That make sense now. Sorry, my bad, I was trying to use it for the Raspberry Pi. I don't know if this is another platform that you would like to support. Appreciated you guys' work.

peterhinch, 2016-09-19:
I have no plans to support the Pi: my interest is in platforms running MicroPython on bare metal. Unless @turbinenreiter has such intentions you may need to look elsewhere, or DIY. It shouldn't be too difficult, but you'd need to study and compare the Pi's I/O with that of the Pyboard.

turbinenreiter, 2016-09-19:
@CharlesCCC 
Step 1: Change the whole driver to take an I2C object as argument instead of creating it itself in imu.py, mpu9150 and mpu9250.py.
Step 2: In imu.py change the _read and _write methods to whatever your platforms equivalent of the contained commands are.
Step 3: Send a pull request.

This should really be straightforward.
Find the RPi I2C doc here: http://www.raspberry-projects.com/pi/programming-in-python/i2c-programming-in-python/using-the-i2c-interface-2

Step 1 is something I wanted to do for the longest time, but I switched using the BMX055 and therefore this is quite low on my list.

peterhinch, 2016-09-19:
@turbinenreiter the driver can already take an I2C object as an argument. We implemented this on 19th March this year.

closed 2017-05-25 by turbinenreiter