micropython-IMU/micropython-mpu9x50 #5: Lopy Migration

gesaleh, 2017-03-15:
Hi
I have done the Lopy porting on my fork but i see acceleration at zero values all the time 
any suggestion ?

peterhinch, 2017-03-16:
I don't have a LoPy so alas I can't test. Do the gyro and magnetometer produce reasonable output?

gesaleh, 2017-03-16:
yes they do 
I have did a fork of you repo and put my modifications
mainly i think the issue can come from the I2C X, Y doesnt work with micropyhton of lopy had to create my I2C object before

gesaleh, 2017-03-16:
you can close the issue there is no issue I can values today i hooked everything from scratch and my modification works I see all values
``
Interrupt: [-0.3061523, -0.9460449, -0.02661133] [-0.7557252, 0.6030535, 1.015267] [36.73865, 67.54931, 22.93968]
Normal:    (-0.3044434, -0.9423828, -0.02807617) (-0.9083969, 0.6946565, 0.9007633) (35.33203, 68.95957, 23.95899)

Interrupt: [-0.3044434, -0.9423828, -0.02807617] [-0.9083969, 0.6946565, 0.9007633] [35.33239, 68.96027, 23.95922]
Normal:    (-0.3046875, -0.9428711, -0.02026367) (-0.778626, 0.740458, 1.152672) (35.15625, 68.43048, 21.41016)
``

peterhinch, 2017-03-17:
OK, I'm glad it's working for you.

closed 2017-03-17 by peterhinch