micropython-IMU/micropython-mpu9x50 #12: "MemoryError:" when trying to run it on an ESP8266 NodeMCU breakout

Thalaivar, 2017-10-11:
How can I get over this error?

peterhinch, 2017-10-12:
The approach to running large modules on RAM limited devices is to use frozen bytecode. Put the MicropPython modules into `ports/esp8266/modules` in the source tree. Then rebuild the firmware and install. This causes the MicroPython code to be compiled to bytecode on the PC; on installation the bytecode is stored in Flash.

Thalaivar, 2017-10-12:
Thanks for your reply! However I'm a newb when it comes to uPy. Till now, to install the firmware, I've been directly using esptool.py and the bin files from the micropython downloads page. How do I rebuild and install the firmware as you mentioned?

peterhinch, 2017-10-12:
A starting point might be to read [the build document](https://github.com/micropython/micropython/blob/master/ports/esp8266/README.md).

General questions are best raised in [the forum](https://forum.micropython.org/). GitHub issues are intended for bug reports.

closed 2017-10-12 by peterhinch