micropython-IMU/micropython-mpu9x50 #14: Adding support for FIFO buffer

jcalbert, 2018-09-05:
The MPU family has a built-in 1024 FIFO which can be set up to read in data from the thermometer, gyroscopes, and accelerometer.

I've been working on building support for that feature, but don't know the ins and outs of memory management or this module yet, so I'd figured I'd mention the project here in case anyone wants to help.

Working branch [here](https://github.com/jcalbert/micropython-mpu9x50/tree/add_fifo).

jcalbert, 2018-09-05:
Note:  I only have an MP6050 to work with, so for now that's all the code supports (as far as I know).

peterhinch, 2018-09-05:
What are the intended applications for this? Much IMU use is as an element in a control system where delays are anathema: the application must read the device at a high rate with minimal latency.

jcalbert, 2018-09-05:
I can see a few uses.

First, Invensense suggests it can be used to reduce power consumption by letting the main board sleep while the MPU is gathering data.

In my case - I'm working on a system that needs to report data leading up to an event (a crash) and it's easier to use the MPU's built-in buffer than program my own to run on the microcontroller.

peterhinch, 2018-09-06:
Thanks for that, I imagine that's exactly the kind of scenario for which it was designed.

@turbinenreiter What's your view on this?

My feeling is that it's rather specialist. This is the first time a user has wanted this, so I'd be reluctant to increase the complexity of our code. But if @jcalbert is happy that it's working and is prepared to maintain and document it, we could put a link to his fork in the docs.

@jcalbert Would you be happy with this approach?

turbinenreiter, 2018-09-06:
Easiest way is to send a merge request. That way, the changes become obvious and whether we merge it or not, it's then part of this GitLab repository.

Generally, I also err on the side of merging. At some point I will have a day off to streamline again.

jcalbert, 2018-09-07:
I'm fine with either a link or a PR once I test a bit more.  As for keeping the code clean, I now think that it might make more sense to move the FIFO code into another another module and import from `imu.py`

jonnor, 2024-03-28:
@jcalbert were you able to make the FIFO code work?