micropython-IMU/micropython-mpu9x50 #15: Random KeyboardInterrupt and ASCII disco

turbinenreiter, 2018-09-05:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "main.py", line 43, in run
  File "/lib/uasyncio/core.py", line 155, in run_forever
  File "/lib/uasyncio/core.py", line 110, in run_forever
  File "main.py", line 33, in send_data
  File "main.py", line 24, in get_data
  File "vector3d.py", line 103, in xyz
  File "imu.py", line 395, in _accel_callback
  File "imu.py", line 312, in accel_range
KeyboardInterrupt: 
>>> KK3K3K<lKlKlKKK3K3K<lKlKlKK<3K3K3K3lK<KK<KK3K3K<lKxKlKKK3K3K<lKlKlKK<,K3K3K3KlK<KK<KK3K3K<lKKHKK3KK<lKlKlKK<,K3K3K3KlK<KK<KK3K3K<lKlKlKK3K3K<lKlK
```

So this is a fun little thing. I'm running on an ESP32 and have a fast loop calling all sensor values of the MPU9265 (which is yet another version of the chip). Every once in a while, execution stops with above Traceback.

Anyone ever seen anything like this?

The line in the code where this happens is:

```
@property
def accel_range(self):
 """
Accelerometer range
Value:              0   1   2   3
for range +/-:      2   4   8   16  g
 """
    try:
        self._read(self.buf1, 0x1C, self.mpu_addr)
        ari = self.buf1[0] // 8 # this line throws up
    except OSError:
        raise MPUException(self._I2Cerror)

    return ari
```

I think this is gonna be a fun one.

turbinenreiter, 2018-09-05:
Reducing the I2C frequency seems to help.

closed 2018-09-05 by turbinenreiter

peterhinch, 2018-09-05:
That is bizarre. I've never seen anything like it.