micropython-IMU/micropython-mpu9x50 #17: Random value from yaw calculation?

eddyyanto, 2020-04-16:
I am using MPU-9250/6500 and managed to get the filtered pitch and roll data just fine. But when I tried to calculate the yaw using the below yaw calculation at https://engineering.stackexchange.com/questions/3348/calculating-pitch-yaw-and-roll-from-mag-acc-and-gyro-data

`yaw = 180 * atan (accelerationZ/sqrt(accelerationX*accelerationX + accelerationZ*accelerationZ))/M_PI;`

The returned value seemed to be a bit random? Any idea?

peterhinch, 2020-04-16:
I suggest you use [Madgwick fusion](https://github.com/micropython-IMU/micropython-fusion) which has been tested.

eddyyanto, 2020-04-16:
Will do! Thanks @peterhinch! :)

eddyyanto, 2020-04-16:
Tried using Fusion and it seemed that my Adafruit Huzzah ESP8266 can't take the big library size:

`
File "main.py", line 4, in <module>
MemoryError: memory allocation failed, allocating 761 bytes
`

I've cleared up unnecessary files  and it keeps throwing me *memory allocation failed* issue.

peterhinch, 2020-04-16:
You probably need to use [frozen bytecode](http://docs.micropython.org/en/latest/reference/packages.html?highlight=freezing) also [my build scripts](https://github.com/peterhinch/micropython-samples/tree/master/fastbuild) as the official docs aren't entirely up to date.

I don't think I've ever tried sensor fusion on an ESP8266.

eddyyanto, 2020-04-16:
Thanks for the pointer, @peterhinch! 🙏
I'll give it a shot! I'll close this thread now.

closed 2020-04-16 by eddyyanto