micropython-IMU/micropython-mpu9x50 #18: How fast we can read data from the sensor

nursanamar, 2020-05-01:
a lot of example that i saw always using the sleep function on each reading, is it possible to not using the sleep function?, and if it is necessary to use, what is the shortest time to sleep?. 

BTW, I am using mp6050 with class MPU6050 in imu.py file

peterhinch, 2020-05-02:
The delays are largely to limit the rate at which data scrolls off the screen. The maximum rates that make sense depend on the low pass filter settings you choose. See [the docs](https://github.com/micropython-IMU/micropython-mpu9x50/blob/master/README_MPU9150.md).

nursanamar, 2020-05-02:
Thanks for the response!

So, by making the `filter_range` lower give me a lower delay but  also increased the noise, right?

peterhinch, 2020-05-02:
Yes. You may need to consider the effects of vibration which (depending on application) can be huge.

nursanamar, 2020-05-02:
Thanks for your time I really appreciate it.

Closing the issue

closed 2020-05-02 by nursanamar