micropython-IMU/micropython-mpu9x50 #19: i can't use mpu6050 by ur module as the demo.

Irvingao, 2020-11-30:
[image] 
i had tested the the code:
`from imu import MPU6050

imu = MPU6050('Y')
print(imu.accel.xyz)
print(imu.gyro.xyz)
print(imu.temperature)
print(imu.accel.z)`
i connnected "X9" and "X10" for "SCL" and ""SDA".
but always :MPUException: I2C failure when communicating with IMU.
Thank you for help!

peterhinch, 2020-11-30:
The 'Y' side I2C connection is on Y9 and Y10. See [the docs](http://micropython.org/resources/pybv11-pinout.jpg).

Irvingao, 2020-11-30:
but i still met the same problem that" I2C failure when communicating with IMU". how can i do next?

Irvingao, 2020-11-30:
and when it inits, i have to wait for over 30 seconds, it that too slow
?

Irvingao, 2020-11-30:
and when it inits, i have to wait for over 30 seconds, it that too slow?

peterhinch, 2020-11-30:
I can't explain that behaviour. Initialisation is normally immediate. I wonder if you have a genuine InvenSense chip or a Chinese clone?

peterhinch, 2021-03-19:
Closing this. I assume it's resolved.

closed 2021-03-19 by peterhinch