micropython-IMU/micropython-mpu9x50 #20: MPU6050 onboard DMP (feature request)

fanlessfan, 2021-03-10:
Hello,

I found a Arduino library which can read from the onboard DMP. Could anyone who has the knowledge to convert it to mycropython?

https://github.com/jrowberg/i2cdevlib

thx

peterhinch, 2021-03-11:
The DMP is proprietary and undocumented, which is why we never attempted to support it. There are two solutions, firstly our sensor fusion library. Secondly the BNo055 module which performs sensor fusion internally in a publicly documented manner. This works exceptionally well and would be my suggestion for any new project.

While I applaud the ingenuity of the author in reverse engineering DMP, I think the effort involved in porting this would be of limited interest given the above existing solutions. Another issue is that sensor fusion in a 6-DOF IMU is of limited use: to do a worthwhile job would require including MPU9150 and MPU9250.

I therefore won't be taking this up. We'll see if other members are interested.

peterhinch, 2021-03-19:
Closing this: I don't think anyone is taking this up.

closed 2021-03-19 by peterhinch